PubInv/covid19-vent-list #10: Please add: DIY Beatmungsgerät

sbamueller, 2020-03-25:
Please add: https://devpost.com/software/diy-beatmungsgerat
We are currently exploring options and submited this design for the #WeVsVirus Hackathon of the German Goverment.

RobertLRead, 2020-03-25:
Thank you. After studying your website, I have added your project as the highest scoring on the spreadsheet and listed it first among buildable projects. Well done! Let us hope and pray you succeed. I would like to see some information about if you have tested it for 48 hours, and more information about monitoring and alarming as per the UK standard (https://assets.publishing.service.gov.uk/government/uploads/system/uploads/attachment_data/file/874279/RMVS001_Rapidly_Manufactured_Ventilator_Specification__PDF.pdf).  Please forgive me if you have that and I missed it (you know the jokes about Americans only speaking English!)

closed 2020-03-25 by RobertLRead

sbamueller, 2020-03-25:
We have not built a design and not yet tested, but hope to do so during this week. It’s all going very fast. There will be a longer article in the German make magazine tomorrow I guess, outlining our approach. 

Sebastian Müller