PubInv/covid19-vent-list #16: OpenLung BVM Ventilator link

Fearfulbot, 2020-03-28:
Link to the main of OpenLung BVM Ventilator GitHub isn't the one in Spanish. 

https://gitlab.com/open-source-ventilator/OpenLung
(Listed as the E-vent from MIT in the spreadsheet)

They're based in Ireland and have a 2k member slack group that's worth including. It seems that they've closed invitations, but can probably sign up through their site: https://opensourceventilator.ie

Thank you!

RobertLRead, 2020-03-28:
Can you check again? If this is still true please give me precise
instructions. It looks good to me right now---my colleagues were just
working a quite a bit on it, however, you may have seen it in a temporary
condition.

Fearfulbot, 2020-03-28:
Oh yes the slack group is still very much in operation, there's just not an open invitation link where people can join freely. I don't quite know what the process is to request invitations.

Regardless, the link listed under OpenLung BVM Ventilator on the spreadsheet goes to a gitlab that's in Spanish, the correct link is: https://gitlab.com/open-source-ventilator/OpenLung for this project. 

[image]

RobertLRead, 2020-03-30:
fixed

closed 2020-03-30 by RobertLRead