PubInv/covid19-vent-list #20: Cuirass type ventilator

jps2000, 2020-03-30:
here is another link you may like to add
https://github.com/jps2000/Cuirass-Ventilator

RobertLRead, 2020-03-30:
I have added it. It is difficult for me to evaluate clinical suitablity. You do not have a license  statement; may I suggest CC0 for desgins and GPL 3 for code? Get back to me if you make a change. Thanks for your efforts!

RobertLRead, 2020-03-30:
Would you like to offer an email for us to add to our public spreadsheet as a point-of-contact?

closed 2020-03-30 by RobertLRead

jps2000, 2020-03-30:
point of contact should be handled via https://github.com/jps2000/Cuirass-Ventilator / Issues
License added

jps2000, 2020-04-04:
@ RobertLRead 
Just added a cuirass construction  designed with F360. f3d File is included  !!!!!