PubInv/covid19-vent-list #21: Open Source Ventilator Ireland

synergiator, 2020-03-30:
Latest update from March 28 2020 introduces a 1 point score drop, from 1.57 to 0.57.

Questions:

- how to know why a score changes? are results of an evaluation run-through published somewhere?
- is it possible to introduce a score delta column in the evaluation?
- is there an open community process to discuss the evaluation criteria (which already are quite comprehensive but maybe other experts could make valuable suggestions)?

RobertLRead, 2020-03-30:
Good questions.

The scores are done by me and an assistant. We do not check projects until a project informs us of something that might change a score, or we happen to stumble upon it and notice that it has changed.

We could introduce a delta score, but I don't see enough value in doing that. It is a challenge keeping up with things as they are.

Making an open process for changing or reviewing the criteria is probably a good idea. I will post a "commentable" link to the Google Doc so people can comment on it. Thanks for the suggestion!

I will leave a comment.

As to why a score dropped or raised, sometimes we make mistakes and rethink things; I hope nobody will be offended. I will re-evaluate any score in response to somebody asking. In this case, I don't remember.

RobertLRead, 2020-03-30:
I added this:
Each of the projects is evaluated on 5-point scale defined here on each of these attributes. Uses this [link](Each of the projects is evaluated on 5-point scale defined here on each of these attributes. Uses this link to comment on those attributes.) to comment on those attributes.

RobertLRead, 2020-03-30:
You can comment on the critera here: [link](https://docs.google.com/document/d/1lWXCij-7yrbWUJJx-y4d6dREdYaTkb62iOZDTDbRi0I/edit?usp=sharing), which as been added to the front page.

synergiator, 2020-03-30:
thank you so much!

Without being offended, we can see that for OSV the field "openness" has been reduced from 5 to a lower value (we have GitLab and to my best knowledge nothing changed here). In case that's an error please update.

synergiator, 2020-03-30:
@RobertLRead I have discussed this also with others, there has been a suggestion to put the eval list on GitHub as well.

RobertLRead, 2020-03-30:
You mean the link to the document is not sufficient?

RobertLRead, 2020-03-30:
I have changed your score to 4 for openness, I may not have seen the GitHub link at the bottom previously, sorry.

closed 2020-03-30 by RobertLRead

synergiator, 2020-03-31:
^^ I am personally happy with the Googledoc, the request for GitRepo came from a peer.

RobertLRead, 2020-03-31:
So you want the eval criteria to be a text file in the repo?

synergiator, 2020-03-31:
if it's possible, would be great, yes. So far, still most engineers focus on actually building something working but I think from the 1001 ventilator projects some will come up and build a powerful upstream which also will need your team. Also if there will be other open test labs.

Also you could clearly state the license.

RobertLRead, 2020-03-31:
I've added it as a separate file in Markdown. I believe the license is
clearly stated as CC0.