PubInv/covid19-vent-list #22: Introduce data interface to support faster exchange and evaluation

synergiator, 2020-03-30:
If it makes any sense, test labs and product teams could consider to exchange data digitally.

That is, many initiatives attract capable software engineers so it's less a capacity problem, just need to find common base/goal to collaborate.

Implementation approaches could be:

* push - initiatives submit standardized data reports to an API endpoint at your site
* pull - initiatives publish standardized data reports on their sites which you can collect

What seems initially to be a sort of overdoing just to simplify reviewing websites and GitLab repositories, will save time on mid term and on long term if community picks up this topic, it would be possible to exhange large data sets for example also ventilator measurement data - from acceptance testing to medical research. Also, this would add to new ways of medical engineering data corpora.

RobertLRead, 2020-03-30:
Do you mean breath trace data?  Or data about progress? I certainly believe we need an open Breath trace standard (I am working to JSON).

synergiator, 2020-03-31:
I mean in this case data about project progress (so far you seem to keep up very well though). The breath trace data is another important topic! Needs also more stakeholders for that to have a solid foundation in terms of common sense from different viewpoints I think. But anyway initial ideations in this area will be picked up, if there has not ever been such a standard.

Regardining my comment in the another issue, I did not notice the CC0 disclaimer in the Googledoc - please take my apologies.

closed 2020-04-13 by RobertLRead