PubInv/covid19-vent-list #27: Ventilator project based on common 3d printer hardware

Vroumdev, 2020-03-31:
Working prototype using Ambu bag.
Tested the prototype at 30 BPM with 37cm H2O (size of the jerrican).  
Website : https://20100.be/ad-hoc-ventilator/

A work in progress

RobertLRead, 2020-03-31:
Added.  If it makes major improvements, send me a note and I will update it in the README listing and spreadsheet.

closed 2020-03-31 by RobertLRead

Vroumdev, 2020-04-03:
Thank you for your work, really useful and motivating.

Here are the last improvements :

-New video with LCD interface and demonstration of different parameters.
-Works standalone without computer control.
-Added LCD user friendly interface to parameters : BPM, Tidal volume (not yet calibrated) and I/E ratio. Parameters taken into account when ventilating.
-Can be started/stopped using LCD interface.
-Interface can handle multiple languages.
-Added Emergency stop (LCD) menu for the moment.
-Tested sound alarm.
-Contact form on website.

Best regards,
Vincent Roumieux

RobertLRead, 2020-04-03:
Thanks. I increased COVID-19 suitability and clinician friendliness.