PubInv/covid19-vent-list #28: column request: BOM availability as column criteria

EnginerdingDan, 2020-04-01:
Would you consider adding a column to your spreadsheet titled "BOM Available".

To indicate whether a project has completed that very important milestone or having a parts list?

In my humblest of opinions, it would be an invaluable resource for people with electronics experience to find the most advanced projects. Thank you.

P.s. your project is absolutely phenomenal! Amazing work.

RobertLRead, 2020-04-01:
I've asked Enrique to think about it.  I don't want it as a separate yes/no
boolean. But it should be part of the buildability spec that we did not
call out explicitly. Possibly we should create "manufacturable" as distinct
from "buildable" and it would go there.

EnginerdingDan, 2020-04-01:
Happy to do as much "grunt work" on this also, since I'll be doing it either way.  Thanks.

RobertLRead, 2020-04-01:
EngineeringDan, why don't you work with my colleague Enrique (Juan). I'd
like the two of you to come up with a 5-point scale to represent "readiness
for manufacture". This will be different than "buildability".
"Buildability" is the ability of one team to build a prototype based on
published documents. "Manufacturability" encompasses things like having a
BOM, having reasonable expectation of the supply chain, having instructions
that are clear enough a semi-skilled worker can execute them (as opposed to
a lead engineer/hobbyist.)

EnginerdingDan, 2020-04-01:
OK, I'm in.   

@Enrique, what are your thoughts on grading for "Readiness
for Manufacture" (also, my vote would be just to title it "Bill of Materials" or "Parts List" for simplicity)

In my estimation, the presence of a bill of materials is a major sign of project progress and viability.  We could rate on a scale like this:

0:  No Bill of Materials found.
1:  BOM present, but some major components not listed.
2:  BOM present, all major components listed, and some minor ("jellybean") components listed.
3:  BOM present, all major and minor components listed, some w/ prices. 
4:  BOM present, all major and minor components listed w/ prices. 
5:  BOM present, all major and components listed w/ prices.  An attempt has been made for substitutions in the developing world  to lower cost, or use common commercial or household items.

Hope this gives us a good start.   Cheers.

RobertLRead, 2020-04-01:
Let's add to those levels other things you should have for a
replicable design, not purely the BOM.

for example, at the Level 5 lets have "investigation into quantities
available in supply chain in 100s or 1000s" and at Level 4 electrical
schematics and mechanical/cad designs of all enclosures and custom parts

etc.  The idea here to capture how hard it would be for someone to set up a
line to manufacture 1000 of these.

RobertLRead, 2020-04-04:
Here is my first cut. Feel free to put comments in the Google Doc:

Manufacturability (1000s)
Note: This is zero until Buildability of (1) unit (separate column) reaches 5.

 0. Insufficient plans to duplicate a single unit.
1. Bill of Materials (BOM) clear.
2. 2d parts, 3d parts, code, all clearly documented.
3. Electrical schematics and air circuit schematics clear. PCBs if any present and documented. Wiring if required fully documented.
4. Basic instructions and special instructions present. Video instructions helpful. Documented in language of choice, preferable more than one. Basic description of “smoke tests” and simple quality assurance present.
5. Either evidence of BOM availability in units of 1000, or supply-chain flexibility of parts suggesting same. Documentation for handling supply-chain disruption present if minimal. Plans include manufacturing issues for non-expert workers. Detailed quality assurance plans present.

RobertLRead, 2020-04-06:
Dan, can you comment on that? I'd like to close this issue.  It is not quite what you asked for, but hopefully it overlaps it enough.

closed 2020-04-13 by RobertLRead