PubInv/covid19-vent-list #33: Add Medtronics...

RobertLRead, 2020-04-01:
Originally Medtronics release was not very open; I have heard it now has much more material.  I would love someone to evaluate it as an opinion and leave it in this issue.

synergiator, 2020-04-01:
An overview of package contents.

Level 1: what's inside the archive? (229 files, with some ZIP archives)

`
./Electronic PCBAs
./Electronic PCBAs/GR106768_Buzzer_PCBA
./Electronic PCBAs/GR106768_Buzzer_PCBA/3D_Model
./Electronic PCBAs/GR106768_Buzzer_PCBA/3D_Model/GR106768B00.zip
./Electronic PCBAs/GR106768_Buzzer_PCBA/PCB
./Electronic PCBAs/GR106768_Buzzer_PCBA/PCB/GR106771A00.zip
./Electronic PCBAs/GR106768_Buzzer_PCBA/PCB/GR106771A00.pdf
./Electronic PCBAs/GR106768_Buzzer_PCBA/BOM
./Electronic PCBAs/GR106768_Buzzer_PCBA/BOM/GR106768_Buzzer_PCBA_BOM.xls
./Electronic PCBAs/GR106768_Buzzer_PCBA/PCBA
./Electronic PCBAs/GR106768_Buzzer_PCBA/PCBA/GR106768B00.pdf
./Electronic PCBAs/GR106768_Buzzer_PCBA/PCBA/GR106768B00.zip
./Electronic PCBAs/GR106768_Buzzer_PCBA/10025028E00_BUZZER BOARD REQUIREMENTS SPECIFICATION, PB500 SERIES.doc
./Electronic PCBAs/GR106768_Buzzer_PCBA/SCH
./Electronic PCBAs/GR106768_Buzzer_PCBA/SCH/10125624A00.zip
./Electronic PCBAs/GR106768_Buzzer_PCBA/SCH/10125624A00.pdf
./Electronic PCBAs/10096510_Battery_Connection_PCBA
./Electronic PCBAs/10096510_Battery_Connection_PCBA/3D_Model
./Electronic PCBAs/10096510_Battery_Connection_PCBA/3D_Model/10096510A00.zip
./Electronic PCBAs/10096510_Battery_Connection_PCBA/PCB
./Electronic PCBAs/10096510_Battery_Connection_PCBA/PCB/10096506A00.pdf
./Electronic PCBAs/10096510_Battery_Connection_PCBA/PCB/10096506A00.zip
./Electronic PCBAs/10096510_Battery_Connection_PCBA/BOM
./Electronic PCBAs/10096510_Battery_Connection_PCBA/BOM/10096510_Battery_Connection_PCBA_BOM.xls
./Electronic PCBAs/10096510_Battery_Connection_PCBA/PCBA
./Electronic PCBAs/10096510_Battery_Connection_PCBA/PCBA/10096510A00.PDF
./Electronic PCBAs/10096510_Battery_Connection_PCBA/10024249C00_BATTERY CONNECTION BOARD, US, PB 540 SPECIFICATION .doc
./Electronic PCBAs/10005011_Power Management_PCBA
./Electronic PCBAs/10005011_Power Management_PCBA/10025029E00_POWER BOARD REQUIREMENTS SPECIFICATION, PB 540 .doc
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/PCB
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/PCB/10039642C00_MANUFACTURING FORM_ PRINTED CIRCUIT_PS BOARD TANTALUMLESS.zip
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/BOM
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/BOM/3842800_POWER SUPPLY BOARD - TANTALUM LESS_BOM.xls
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/PCBA
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/PCBA/3842800F00_3D_Model.zip
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/PCBA/3842800F00.pdf
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/SCH
./Electronic PCBAs/10005011_Power Management_PCBA/3842800_POWER SUPPLY BOARD - TANTALUM LESS_PCBA/SCH/10039639C00_Power Supply Board - Tantalum Less_SCH.pdf
./Electronic PCBAs/10106052_PowerPack_PCBA
./Electronic PCBAs/10106052_PowerPack_PCBA/3D_Model
./Electronic PCBAs/10106052_PowerPack_PCBA/3D_Model/10106054A00_3D_Model.zip
./Electronic PCBAs/10106052_PowerPack_PCBA/3D_Model/10106054A00.PDF
./Electronic PCBAs/10106052_PowerPack_PCBA/PCB
./Electronic PCBAs/10106052_PowerPack_PCBA/PCB/10107738A00.zip
./Electronic PCBAs/10106052_PowerPack_PCBA/PCB/10107738A00.pdf
./Electronic PCBAs/10106052_PowerPack_PCBA/BOM
./Electronic PCBAs/10106052_PowerPack_PCBA/BOM/10106052_POWERPACK2 BOARD (DUAL BATTERY)_BOM.xls
./Electronic PCBAs/10106052_PowerPack_PCBA/PCBA
./Electronic PCBAs/10106052_PowerPack_PCBA/PCBA/10106052C00.zip
./Electronic PCBAs/10106052_PowerPack_PCBA/PCBA/10106052C00.PDF
./Electronic PCBAs/10106052_PowerPack_PCBA/10025023G00_BATTERY HARDWARE SPECIFICATION, PB560 PB540 PB520 .doc
./Electronic PCBAs/10106052_PowerPack_PCBA/SCH
./Electronic PCBAs/10106052_PowerPack_PCBA/SCH/10105965B00.DSN
./Electronic PCBAs/10106052_PowerPack_PCBA/SCH/10105965B00.pdf
./Electronic PCBAs/GR106769_CPU_PCBA
./Electronic PCBAs/GR106769_CPU_PCBA/3D_Model
./Electronic PCBAs/GR106769_CPU_PCBA/3D_Model/GR106769B00.zip
./Electronic PCBAs/GR106769_CPU_PCBA/10025025B00_OXYGEN VALVE HARDWARE SPECIFICATION, PB 540 .doc
./Electronic PCBAs/GR106769_CPU_PCBA/PCB
./Electronic PCBAs/GR106769_CPU_PCBA/PCB/GR106772A00.zip
./Electronic PCBAs/GR106769_CPU_PCBA/PCB/GR106772A00.pdf
./Electronic PCBAs/GR106769_CPU_PCBA/10025026C00_RELIEF VALVE HARDWARE SPECIFICATION, PB 540.doc
./Electronic PCBAs/GR106769_CPU_PCBA/3832100F00_SOFTWARE REFERENCE DRAWING PB540 POWER SUPPLY.doc
./Electronic PCBAs/GR106769_CPU_PCBA/10025024C00_EXHALATION CONTROL VALVE HARDWARE SPECIFICATION, PB 540.doc
[... 951 lines cut ...]
  Length      Date    Time    Name
---------  ---------- -----   ----
    19545  2012-11-02 12:07   3810200A00.PDF
   908197  2012-08-03 09:40   38102000300.pdf
   159232  2012-04-09 18:43   3810200 butВe carte commutation.SLDPRT
   632320  2012-11-02 12:07   3810200.SLDDRW
    43532  2012-11-02 12:07   3810200_Approved_Revision_03.pdf
---------                     -------
  1762826                     5 files
Archive:  ./Mechanical Part Drawings/3825500/3825500E00.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
  4665856  2018-01-25 13:46   3825500.SLDPRT
 17953792  2018-01-25 13:47   3825500.slddrw
---------                     -------
 22619648                     2 files
Archive:  ./Mechanical Part Drawings/10000915/10000915B00.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
   752128  2012-07-24 12:33   10000915.SLDPRT
   616960  2012-07-26 14:40   10000915.SLDDRW
---------                     -------
  1369088                     2 files

`

RobertLRead, 2020-04-01:
[image]

RobertLRead, 2020-04-01:
Will adjust as the situation with software, community support, etc. becomes clearer. Thanks!

RobertLRead, 2020-04-01:
Better screen shot:

[image]

closed 2020-04-01 by RobertLRead

synergiator, 2020-04-01:
there is also one special issue with their license: I have never seen licenses previoisly which are limited by time/events.

This license is valid until 2024 either the pandemic end. So it's a very tentative open source venitlator.

RobertLRead, 2020-04-01:
I'm not concerned about that at all; it is good enough for now. However,
that does not mean other open source projects are not valuable!

drivepilot, 2020-04-02:
They've released the software, sorry about the formatting. 

`Archive:  ./AL020004.zip
  Length     Date   Time    Name
 --------    ----   ----    ----
        0  03-31-20 09:49   AL020004_updated/
        0  03-31-20 12:28   AL020004_updated/Sources/
        0  03-31-20 09:49   AL020004_updated/Sources/Config_MPLAB/
     1955  03-30-20 01:08   AL020004_updated/Sources/Config_MPLAB/18f6622_e_Alim_Legendair_Phase_II.lkr
    17407  03-30-20 00:55   AL020004_updated/Sources/Config_MPLAB/Alim_LEGENDAIR_Phase_II.mcp
   285696  03-30-20 00:55   AL020004_updated/Sources/Config_MPLAB/Alim_LEGENDAIR_Phase_II.mcw
        0  03-31-20 09:49   AL020004_updated/Sources/Md5Sum/
   253952  03-30-20 00:55   AL020004_updated/Sources/Md5Sum/Md5Sum.exe
      512  03-30-20 00:55   AL020004_updated/Sources/Md5Sum/Readme.txt
        0  03-31-20 09:49   AL020004_updated/Sources/scripts/
    27214  03-30-20 00:54   AL020004_updated/Sources/scripts/auto_tu.py
        0  03-31-20 09:49   AL020004_updated/Sources/scripts/auto_tu_templates/
       52  03-30-20 00:55   AL020004_updated/Sources/scripts/auto_tu_templates/metrics_footer.html
     1044  03-30-20 00:54   AL020004_updated/Sources/scripts/auto_tu_templates/metrics_header.html
       52  03-30-20 00:54   AL020004_updated/Sources/scripts/auto_tu_templates/tu_footer.html
     1234  03-30-20 00:54   AL020004_updated/Sources/scripts/auto_tu_templates/tu_header.html
    11295  03-30-20 00:54   AL020004_updated/Sources/scripts/BIN2HEX.EXE
      581  03-30-20 00:54   AL020004_updated/Sources/scripts/generate_md5_al.bat
    11642  03-30-20 00:54   AL020004_updated/Sources/scripts/HEX2BIN.EXE
    49152  03-30-20 00:54   AL020004_updated/Sources/scripts/md5sum.exe
      141  03-30-20 00:54   AL020004_updated/Sources/scripts/post_build.bat
      138  03-30-20 00:55   AL020004_updated/Sources/scripts/readme.txt
  2109399  03-30-20 00:55   AL020004_updated/Sources/scripts/updatecrcal.exe
      424  03-30-20 00:54   AL020004_updated/Sources/scripts/update_crc_al.bat
        0  03-31-20 09:49   AL020004_updated/Sources/SRC/
        0  03-31-20 17:44   AL020004_updated/Sources/SRC/ALIM/
    13984  03-31-20 17:44   AL020004_updated/Sources/SRC/ALIM/bootload.asm
     1786  03-31-20 13:27   AL020004_updated/Sources/SRC/ALIM/constant.h
    19918  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/define.h
     7827  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/enum.h
     5879  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/IO_Declare.h
    41428  03-30-20 01:06   AL020004_updated/Sources/SRC/ALIM/p18f6622.h
     2734  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/Saturation.c
     4894  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/STARTUP_Alim_Legendair_Phase_II.c
     2054  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/structure.h
     1718  03-30-20 01:07   AL020004_updated/Sources/SRC/ALIM/typedef.h
        0  03-31-20 09:49   AL020004_updated/Sources/SRC/CAN/
     5721  03-30-20 01:07   AL020004_updated/Sources/SRC/CAN/CAN_Channel_Acquisition.c
     2269  03-30-20 01:07   AL020004_updated/Sources/SRC/CAN/CAN_Channel_Acquisition.h
        0  03-31-20 09:49   AL020004_updated/Sources/SRC/CHARGE/
     5175  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Charge.c
     2112  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Charge.h
     2141  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Data.h
     5591  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Init.c
     2182  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Init.h
    10382  03-31-20 13:46   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Launch.c
     2289  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Launch.h
     2939  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Loop_Controller.c
     1897  03-31-20 16:51   AL020004_updated/Sources/SRC/CHARGE/CHARGE_Loop_Controller.h
     2408  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_StopCharge.c
     1767  03-30-20 01:07   AL020004_updated/Sources/SRC/CHARGE/CHARGE_StopCharge.h
        0  03-31-20 09:49   AL020004_updated/Sources/SRC/DATABASE/
     2875  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Compute.h
     2077  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Compute_Read.c
     2289  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Compute_Write.c
     3335  03-31-20 16:52   AL020004_updated/Sources/SRC/DATABASE/DB_Config.h
     2069  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Config_Read.c
     2296  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Config_Write.c
     3117  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Control.h
     2077  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Control_Read.c
     2303  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Control_Write.c
     4486  03-31-20 13:49   AL020004_updated/Sources/SRC/DATABASE/DB_EEPROM_BAT.h
     1972  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_EEPROM_BAT_Read.c
     2229  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_EEPROM_BAT_Write.c
     4098  03-30-20 01:07   AL020004_updated/Sources/SRC/DATABASE/DB_Measurement.h
[... 1699 lines cut ...]
     1935  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Setting_Adjust_Vsimv_Mode_Ti_Control.h
     3901  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Setting_Adjust_Vsimv_Mode_Vol_Control.c
[... cut ...]
     4221  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Write_Adjust_Vsimv_Mode_Peep.c
     4540  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Write_Adjust_Vsimv_Mode_P_Support.c
     3075  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Write_Adjust_Vsimv_Mode_Ti_Control.c
     6629  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Write_Adjust_Vsimv_Mode_Vol_Control.c
     1817  03-29-20 23:59   LX010106/SRC/VSIMV_SETTINGS/DB_IHM_Write_Adjust_Vsimv_Mode_Vol_Control.h
 --------                   -------
 31516698                   1536 files`