PubInv/covid19-vent-list #37: Evaluate also controller code bases

synergiator, 2020-04-02:
Code base introduce their own complexities; it would be great to have an additional column to grade code quality.

There are varieties of tools, we could start with simply linting to get an idea about a statc code quality.

I am open to discuss further details and implementation in this issue thread.

For example:
0. Code does not even compile, do not need to check static quality.
1. Lots of warnings > 10
2. Some warnings < 10
3. Several non-critical warnings < 5
4. 1 or 2 minor warnings.
5. Zero warnings, 100% code quality.

Further metric could be code coverage whilee 100% code coverage  is very expensive and rarely seen.

Further inspirations for graded code quality (a commercial tool but they have a great quality framework and also many industrial customers; I have reached out to their CEO in this context):
https://www.tiobe.com/tqi/awards/

RobertLRead, 2020-04-03:
This is a good idea. I don't think I would focus on warnings, and even code quality might be narrow...but the overall software capability and flexibility is something worth looking at. Let me talk to my colleague Enrique about this.

RobertLRead, 2020-04-06:
I have added a column (far to the right) for notes on Software.  I'm not ready to start scoring software; as projects become more advanced, I might develop some criteria. Thanks for the suggestion.

closed 2020-04-06 by RobertLRead

synergiator, 2020-04-20:
I think, beyond academic interests, this quick and simple check can be reduced to most significant question:

"is there a transparent automation aka CI/CD to create controller flash file from sources?"

Implications:
* if yes, a test team can download the file and flash the board, or easily recompile the file if needed
* if not, there is an effort to install development IDE and risks that the code would not compile either more time is needed for resolving dependencies.

The CI/CD automation is integral part of sites like GitHub and GitLab so the check should be easy.