PubInv/covid19-vent-list #50: Please update Open Breath entry

simone12345, 2020-04-04:
Hello,

I am a member of the Open Breath team, I would like our project to be reevaluated. last time it was assessed (March 21st) we were at a very early stage.  It is now a very open project, with many members actively contributing to the forum, wiki, and repo.

Thanks

RobertLRead, 2020-04-05:
Thank you. I have re-evaluated it. It remains hard for me to find some
information, such as what the controls look like, and whether you have
alarms and monitors, and what sensing you are going in the device as
opposed to separated testing. Forgive me if I have gotten something wrong.

closed 2020-04-05 by RobertLRead

simone12345, 2020-04-06:
Robert I am sorry I've just answered now, there is a complete block diagram that describes user interface, the alarms and the redundant interface system (we plan to have a more basic LCD system attached to the machine and a remote dashboard).

simone12345, 2020-04-06:
Just to mention our project was built with mass production in mind, it is not a 3D printed devices, since the mechanical part has to be produced by workshops through simple techniques and materials, we are also planning to provide an alternative 3D model for FDM 3D Printed parts soon.