PubInv/covid19-vent-list #51: Army-Type Pneumatic Ventilator

kurtbd, 2020-04-04:
When reviewing your spreadsheet [Analysis of Open Source COVID-19 Pandemic Ventilator Projects](https://docs.google.com/spreadsheets/u/1/d/e/2PACX-1vTYAfldxoIiO46VAWH1NlhrwFBn9mguqS2bh1spnLEu4AVVN1cj1vaEm6vOp5Z6UnaAbUwd8dslCXdM/pubhtml#) I noticed that there was a project called project Army-Type Pneumatic Ventilator. 

I am helping coordinate this project based on these designs and you can see our initiative here >> https://www.worldwideventilator.com/

Can I just go into the google document and add our website and additional information on this effort?

RobertLRead, 2020-04-05:
You can go into the spreadsheet and make comments and suggestions; I hope I
have not published an editable link; if I have let me know. If you add
contact information and a URL (in a comment) we will evaluate it when we
see, or you can send that to me.

kurtbd, 2020-04-05:
I believe this link is editable. https://docs.google.com/document/d/1hRfHGBVZNBeAJTYrPazY6ElMNkyrgIt9aFnz_eUzQG0/edit#heading=h.zd4mdan4zmci

I added our website there. 
https://www.worldwideventilator.com/about-us/

kurtbd, 2020-04-05:
Website: https://www.worldwideventilator.com
Contact Email: Kurt@Freshconsulting.com 
Drive: Pressure 
Notes: A low-cost and open source field emergency ventilator for the world. Currently looking for medical testing help, resources, and collaboration. 
Language: English
Microcontroller: No electronics.

kurtbd, 2020-04-05:
Project Name: Worldwide Ventilator

RobertLRead, 2020-04-05:
I don't believe you have an editable link.  but I will add it.

closed 2020-04-06 by RobertLRead