PubInv/covid19-vent-list #52: Demand-modeling subteam progress and product

SEE-ICT, 2020-04-05:
I'm working with a team in the Santa Clara U datavscovid hackathon, and modeling of ventilator demand is very important to our effort. Where could we find the results of the work of the demand-modeling subteam mentioned in this medium article: https://medium.com/@brucefenton/ventilator-project-update-march-21th-2020-bd2ef9d587e0 ?

RobertLRead, 2020-04-05:
Send your email to me <read.robert@gmail.com> and I will introduce you them
by email.

SEE-ICT, 2020-04-05:
mglazer@see-advisors.com

We’re developing a tool to match those who need vemtilators with those who have them with a particular emphasis on coordinating with suppliers of. improvised ventilators.  

I’m also leading one and involved with another improvised ventilator project in Southeast Europe, and I’ll be speaking to my colleagues this morning our time about getting on your list and using your evaluation criteria. We’ve ben looking for something like your effort for a while now. Has Princeton U’s team found it’s way to you? If not, I’ll make the connection.

RobertLRead, 2020-04-05:
Great! I have not heard from Princeton.

RobertLRead, 2020-04-27:
Thanks for this, but I' not sure what further action can be taken. Get back to me if you need something.

closed 2020-04-27 by RobertLRead