PubInv/covid19-vent-list #62: Non-electric ventilator design

jety89, 2020-04-08:
Pretty awesome, pressure-operated valves with some tubing.
https://www.youtube.com/watch?v=y_aHpr81ASk

Please indicate if this is not the appropriate channel to share.

jety89, 2020-04-08:
https://www.vortran.com/go2vent

https://www.youtube.com/watch?v=afFWwEcIBW4
VORTRAN Automatic Resuscitator (GO2VENT) Training
Vortran Medical Technology 1, Inc.

jety89, 2020-04-08:
Looks like the closest analog is the gtech open-source ventilator

jevillacres, 2020-04-08:
Thanks for sharing! I added the device to the spreadsheet - although it is not open source, i've seen one other group that based their design off of it. Here's the link in case you're interested: https://docs.google.com/document/d/1R25O2mKT3TfSjXTmheGEevIk6rTJ49o-sGCJU3QqP3M/edit#

closed 2020-04-08 by jevillacres

jety89, 2020-04-09:
Awesome