PubInv/covid19-vent-list #74: A recent development in the ventilators story

jety89, 2020-04-11:
Something to be aware of

https://www.youtube.com/watch?v=ar5IYxznmi0
Doctors Say Ventilators Killing THOUSANDS, Will Stop Using Them
The Red Elephants Vincent James

jety89, 2020-04-11:
The issue in this case seems to be wrong settings: inspirational pressure too high, leading to lung damage, somehow leading to some doctors refusing to use them altogether

RobertLRead, 2020-04-11:
Thanks, I've read about the differences in opinion on this. I believe
doctors are successfully learning how to treat the disease better as they
gain more experience. I don't doubt these  results.

However, I view it as the job of the engineering community to give doctors
a full box of tools, not to tell them which tool to use. For the time
being, I believe the open-source ventilator effort remains impactufl.

RobertLRead, 2020-04-11:
Right -- and obviously, any DIY design has to be carefully tested and allow
careful control for precisely that reason.

jety89, 2020-04-11:
Thanks for the feedback

jety89, 2020-04-24:
Some further development/debate
https://www.youtube.com/watch?v=Fz2gyhto-iI
The Great Ventilator Fiasco of COVID19 | Are They Even Needed? Doctor Explains
•Apr 24, 2020
Medlife Crisis

The gist of it is.. some insights from a medical professional.

RobertLRead, 2020-04-27:
Okay, this was valuable but I'm closing this issue now, thank you.

closed 2020-04-27 by RobertLRead