PubInv/covid19-vent-list #76: respiArt ventilator project

artur-iluk, 2020-04-12:
A potential candidate to add - my project respiArt on github.

https://github.com/artur-iluk/respiArt

Prototype is working. in README ther is a  description and link to youtube movie.

Artur Iluk

closed 2020-04-12 by avinashbaskaran

artur-iluk, 2020-04-13:
I'm new on github, and I don't know exactly how is works.
Could you explain me, what does it mean what issue is closed? It was
mentioned, that new projects can be suggested by creating new issue.
Closing the issue means that project will not be included on the list? Why?

Best regards
Artur

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

avinashbaskaran, 2020-04-13:
Artur,

Thank you very much for reaching out to clarify! Your respiArt device was added to the list (currently #30): https://docs.google.com/spreadsheets/d/1inYw5H4RiL0AC_J9vPWzJxXCdlkMLPBRdPgEVKF8DZw/edit?ts=5e8d0ab8#gid=0

I was personally very impressed by its packaging, functionality and user interface. If you release any updates to your project, please add another "issue" requesting re-evaluation. Thanks again for your efforts and good luck.

RobertLRead, 2020-04-13:
An issue is closed when there is no more work to be done. In this case, it
was closed because we added it; in some cases it might be closed the issue
is not actionable.

artur-iluk, 2020-04-13:
Thank you for explanation.
Two remarks:
1. You can put as a contact point my email.
2. There is some community support on Slack.com, but in polish language:
https://app.slack.com/client/T0105JVDQTZ/C0103A4DCBE
A few people is helping me in that project.

I will keep you informed.
Best regards
Artur

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

RobertLRead, 2020-04-13:
If you can give us an English title for your community (the slack team), we
will add that to our page (not the spreadsheet) under the "Communities"
section and note that it is Polish-language oriented.