PubInv/covid19-vent-list #78: order proposals

sbamueller, 2020-04-13:
Order the proposals (maybe using tags into)

- rebuildung an existing device (pb 560, byrd mark 7)
- constructing a new device

which technologies are used
- ambubag based design
- piston pump 
- turbine
- valves using pressure form oxygen/compressed air

RobertLRead, 2020-04-13:
We have "Drive mechanism" column on the right already.  It might look
neater to organize the table that way, but I'm not quite ready to do that.
However, creating something to represent "exising device" or "new device"
is a good idea, I will do that.

RobertLRead, 2020-04-27:
We have added an additional column and analyzed based on "drive" mechanism.  That is not exactly what you asked for, but partially meets the need.

closed 2020-04-27 by RobertLRead