PubInv/covid19-vent-list #124: Re-evelauted the AmboVent since it has been duplicated by an outside team....

RobertLRead, 2020-05-15:
https://www.facebook.com/story.php?story_fbid=10157509286329506&id=565289505&sfnsn=mo

RobertLRead, 2020-05-15:
Dr. Eliram added some additional information:

1. Yes. we do have a few teams who materialized our source code into a real working machine at the prototype level. 

   they include the teams from Guatemala, Canada, USA-Alaska and USA-N.H ,Ukraine and India. they all sent us their video's ( I can share if you wish) 

   there are others in several different stages of construction (we have images only) 

   all and all in our GLOBAL HIVE MIND (WhatsApp) there are 227 members all concurrently busy with their AmboVents.

avinashbaskaran, 2020-05-16:
Amazing progress, happy to re-evaluate

closed 2020-05-16 by avinashbaskaran