PubInv/covid19-vent-list #127: CoroVent updates

tkapler, 2020-05-19:
Hi, i am the manager of the CoroVent project, so i can give you updates for our project
openess, buildability and community support - OK
Reliability testing - 5 - we are testing several ventilators on long term in testing laboratory for more then month without problems, all parts are designed to work 10+ years, except batteries, that are medical grade and certified for >3 years and then will nead yearly check
Suitability - 5 - we meet british standard (and US/Canadian and Russian and CE EU). We have passed EU electrocompatibility tests and we are in standard medical certification processes for all of them (but e.g. CE tests took normaly 1.5 year).
Clinical friendly - 5 - we designed corovent with cooperation with ICU doctors, they are part of the team, so everything was tuned based on their needs and what are they used to. We are even able to do variants for e.g. US market, where some other units are used and some other types of ventilators are common - US doctors are not checking this. Of course the UX is also multilanguage etc.
Manufacturability - 5 - we have designed it especially with this in mind - our manufacturing capacity is 850 units / day, we do have stock (or blocked stock at the vendors) for 5000 units and with the vendors we have agreed monthly capacity of 10-15 000 pcs a month, if there would be enough orders. We are now in contact with EU as we have won the EUvsVirus hackathon in our ventilation category, just this week we are speaking with investors, governments, hospitals ... on EU #matchathon ...

Let me know if you will need some proofs, or more information, we can send you videos, stream from our factory or laboratory etc.

avinashbaskaran, 2020-05-23:
Thank you very much for your request for update, I will try to process this update as soon as possible.
Reliability Testing - Are you generating data from your long term testing and Is the data being reviewed? [5]

closed 2020-05-28 by avinashbaskaran

tkapler, 2020-05-28:
re @avinashbaskaran yep, do you want to send some testing reports etc.? Just let me know (kapler@corovent.cz)