PubInv/covid19-vent-list #128: UPDATE from 59 	Respirador-DQ3D-NICA

DQUEROL, 2020-05-22:
Greetings
I have updated  a series of elements, including continuous mechanical and electonic testing for 8 days and new components. The Respirator should be buildable with mainly locally available parts. Please have look at
https://github.com/DQUEROL/Respirador-DQ3D-NICA/blob/master/Equipo%20apoyo%20respiratorio%20v2.jpg
and
https://github.com/DQUEROL/Respirador-DQ3D-NICA/blob/master/06%20Revisiones%20y%20Testing/Testing.xlsx
and 
https://github.com/DQUEROL/Respirador-DQ3D-NICA/blob/master/06%20Revisiones%20y%20Testing/Pendientes%20y%20avances.docx
Thank you
Daniel Querol

avinashbaskaran, 2020-05-23:
Thank you very much for the update Daniel, would you be able to describe if the device has PEEP capability?

DQUEROL, 2020-05-24:
No PEEP, as i believe it may not be ideal in the case of COVID, and very difficult to implement with the flap valve of the AMBU.
Also, I am aiming at PSV+ or PAV, for asisted respiration of non intubated , pre-ICU patients. 
https://academy.esicm.org/mod/emodulepage/view.php?id=7970

We presently have no ICU beds available in Nicaragua, and we are only at the beginning.
Greetings and stay safe.

avinashbaskaran, 2020-05-27:
Thank you very much for following up with me Daniel, I greatly appreciate your insight. Would you be able to describe any testing your group has done? (ie. testing with a bag, a test lung, a test lung with pressure and volume measurements, etc.)

DQUEROL, 2020-05-27:
I have only tested it on myself, waiting for access to test lung.

Presently working on distal and proximal pressure changes with BPM and 
volumes

greetings
Daniel Querol

avinashbaskaran, 2020-05-28:
Thank you again for the information, Daniel. I've updated the evaluation for your group and am looking forward to seeing your progress. Please reach out to us if you have any further questions

closed 2020-05-28 by avinashbaskaran