PubInv/covid19-vent-list #142: Relationship of projects to open source

paula-stack, 2020-07-09:
Dear colleagues,

Many thanks for maintaining this useful list.

I am writing to ask about the inclusion of the HEV project hev.web.cern.ch

One question we have is if you have a criteria that you only list projects which are fully open source. The spread sheet is entitled "Open Source" but it is not clear from the list if this is a criteria.

If this can be clarified we would like to ask that you consider the inclusion of hev.

Best regards Paula Collins

RobertLRead, 2020-07-09:
We prefer open source but we evaluate projects which are not and will
include yours. However, it is often very difficult to evaluate projects
which are not open source, as the design is harder to understand.

I have been trying to get my volunteers to process these issues for a
while; we are a little behind schedule.

paula-stack, 2020-07-09:
Dear colleagues,

Many thanks for your email and the kind offer!

I agree that open source is a tricky definition and probably there is a wide range of compliance on the list of projects you already have, including some projects that originally declared for open source but are no longer on that track.

Given though that the title of the spread sheet is “Open Source”, we would unfortunately prefer to not include HEV as we don’t want to give the impression that this decision is finalised, at least as far as our project is concerned.

If the project is expanded to include all novel designs, including those who may in the future be open source, we’d be delighted to be included, but then we would ask that the title of the spreadsheet is changed or that this is somehow clarified.

Again, many thanks for your work which I understand is largely volunteer, it’s a fantastic resource and we are lucky in the community that there are people who devote time and effort to this kind of complilation.  If we would be considered for inclusion then we would be happy to supply you the draft of our new arxiv note which will appear in a few days and will give a huge amount of detail which might help the evaluation.

Many thanks,

regards, Paula Collins

avinashbaskaran, 2020-07-09:
Hello Paula,

We will include your design with a (1) ranking for openness, meaning "Not Open"at the present. This may allow for increased awareness and potential support of your efforts via our spreadsheet audience. If your decision regarding openness changes, this can be updated. Please let me know if there are any objections to this course of action

closed 2020-07-09 by avinashbaskaran

paula-stack, 2020-07-09:
Dear Avinash,

Many thanks for your email and proposal.

Unfortunately it is not possible to include HEV in this way but thank you anyway for the proposal.

Best regards Paula