PubInv/covid19-vent-list #162: Please add RED Ventilator to your list

mgurunathan, 2020-08-31:
Hello, we have an emergency ventilator project that I would like to have added to your master spreadsheet.  Our project is called "Respiratory Emergency Device (RED)", or "Project RED".  We are located on the web at https://redventilator.org .

Our project was started in early April and we are a group of engineers who have 5 years of previous ventilator design experience.  We are designing a blower based ventilator which will be manufacturable for under $4K.  It will provide many features seen in more expensive ventilators.  We plan to have a small number of prototypes ready by end of September.  Also, we have formed partnerships with teams in Nepal, India and Argentina who are interested in mass producing our design for their citizens.

Point of contact for the project is:

Mohan Gurunathan
mohan@redventilator.org

Please let me know if you have any questions.  Thanks!

Mohan

avinashbaskaran, 2020-09-19:
Mr. Gurunathan,

Thank you very much for this update, and apologies for our delay in adding this item. We would like to include ranking data regarding the categories available here (https://docs.google.com/document/d/e/2PACX-1vRl9yZ27KvslftcNvweHgH1A81pO8gHL62TWpY_VY-UELWdK9x-4-3hNw3DbkemClzExPsg8RfnxilP/pub), could you please provide some background on your group's device?

closed 2020-09-19 by avinashbaskaran

mgurunathan, 2020-09-20:
Hi @avinashbaskaran ,

Here are my rankings for our project:

Openness
4.	Has a clear license strategy, regular updates to plans

Buildability
4.	Complete documentations suggested reproducibility

Community Support
3.	Responsive leader or manager, more than one volunteer

Functional Testing
 0.  In Design Phase, Not listed/tested

Reliability Testing
 0.   Not Listed

COVID-19 Suitability   
0.  In design phase/Not listed

Clinician Friendly
3.	Breath rate, volume and pressure control easy to set, standard ports, clear external labeling graphically and in the language of choice

EMC (Electromagnetic compatibility) Testing
1.	In Design Phase, Not listed/tested/addressed

National Approval Agency/ EUA approvals
0.   Not listed/tested/addressed

Usage in field
      0.        
1.	Not deployed

Financing
1.	Blocked by need for investment

Manufacturability (1000s)
0. Insufficient plans to duplicate a single unit.

Manufactured Products/ produced products/ produced units
1.	Manufacturing planning in progress 

Please let me know if you require any other information.  Thanks.

Mohan

avinashbaskaran, 2020-09-20:
Thank you very much, Mohan!