PubInv/covid19-vent-list #168: Flow-i Bridge project

aeon-lakes, 2020-12-16:
12 Bridges were manufactured on my behalf by RAM3D in Tauranga for our local use and have been distributed to Rotorua and Gisborne hospitals. However, there is no opportunity to try them out here as there is no COVID in the community in NZ.

RobertLRead, 2020-12-16:
Okay, thank you --- do you have a link to the project?

aeon-lakes, 2020-12-16:
Only online presence is on GrabCAD and the associated discussion group, which has attracted members but no discussion so far. 

https://grabcad.com/library/flow-i-bridge-1 <https://grabcad.com/library/flow-i-bridge-1>

That also has links to a medRxiv paper in the documentation. Which will probably stay unpublished unless someone uses the design and also shares their results.

Our local manufacturer also has a write-up about the project on their webpage here: https://www.ram3d.co.nz/post/ram3d-are-printing-ventilators-parts-for-covid-19-pandemic <https://www.ram3d.co.nz/post/ram3d-are-printing-ventilators-parts-for-covid-19-pandemic>

Made in Ti64 and by RAM3D the Bridge inserts appear eminently usable. The Flow-i anaesthetic machine passes system check with only a small increased leak, and they allow the use of the Flow-i away from the OR with no scavenging (ie. in an ICU) and with humidified ‘wet’ circuits, but without internal water rainout (over 48 hours continuous running with a test lung).

The part has been placed in equipment stores in the two NZ hospitals (8 in Rotorua and 4 in Gisborne) that have  a Maquet fleet, but (thankfully) we haven’t had to use them in anger due to a distinct lack of COVID here. Nor have I been able to test more extensively than I did because we flipped back to trying to catch up with the lockdown backlog of elective work, so the machines have been in use flat out doing what they’re actually designed for.

Also note that EMC testing wasn’t done, as it should be no different to the unmodified Flow-i —all I’ve done is replace some internal plastic in the gas channel with a lump of medical grade titanium.

Financing was provided by Lakes DHB (as they employ me). Engineering work and reverse engineering was a charitable endeavour by Kilwell Fibretube and RAM3D (and as it helped them stay open). 

Anyway, the files are there for others to try and replicate the insert. Or RAM3D will happily print another batch to order for $.

Cheers

Andrew
--
Dr Andrew J Robinson BM FRCA
Consultant Anaesthetist
Rotorua Hospital, Lakes DHB, New Zealand

closed 2021-04-29 by avinashbaskaran